0xCAFEDECAF/VanBus #2: writing packets to the VAN

jnslsrwy, 2020-08-09:
Hi, first of all thank you for this amazing lib! I wanted to ask why you chose to limit the lib to ro from the bus. Is this feature planned for a future update and if so what are the difficulties to overcome if there are any?

0xCAFEDECAF, 2020-08-10:
Hi, writing to the VAN bus is not yet possible because I have not implemented it yet. Maybe in the future, if I have some spare time, I will see what may be possible in that area.

morcibacsi, 2020-08-10:
Writing the VAN bus can be quite challenging. There are more than one message types which also complicates things. In the TSS463 datasheet there is a good explanation on what should be taken care of. The device should listen on the bus for a specific amount of time. When in this time frame the bus is free then it can start transmitting (and in the meantime also listening). If it detects that there's a difference between what it transferred and what it received it must stop transmitting (this is called bus arbitration). Otherwise we overwrite a message that another device is sending.
However I admit that it would be awesome to have a software only solution for this. 
There is something similar to this implemented for a PIC microcontroller here: https://www.cnblogs.com/shangdawei/p/4769620.html

jnslsrwy, 2020-08-11:
> Hi, writing to the VAN bus is not yet possible because I have not implemented it yet. Maybe in the future, if I have some spare time, I will see what may be possible in that area.

Ah I see. Thanks for your reply 👍🏼

> In the TSS463 datasheet there is a good explanation on what should be taken care of.

Will dig into that! Could you give me some other sources to learn more about the VAN standard or is the best way to simply read datasheets?

> There is something similar to this implemented for a PIC microcontroller here: https://www.cnblogs.com/shangdawei/p/4769620.html

I recall that guy Graham Auld doing something with a PIC (http://graham.auld.me.uk/projects/vanbus/van-pioneer.html) but I could not find the source code of his firmware for further research. I believe he did not release it.

morcibacsi, 2020-08-11:
I read a lot of forums and the documents which is mentioned by @0xCAFEDECAF in the readme, but the datasheet of the TSS463 contains everything which is needed to understand the protocol. However I admit that it is not the easiest to read (I read it so many times that I almost know what does it contain on every page). 
I have the source code from Graham but it is also for reading only. It is written in assembly so it didn't help me much. The only publicly available library for writing is probably mine. You should check my repositories for further information. I spent a lot of time on reverse engineering the VAN bus protocol and tried to release everything I figured out.

0xCAFEDECAF, 2020-11-19:
Just uploaded a new version that supports rudimentary packet transmitting functionality.

closed 2020-11-19 by 0xCAFEDECAF