0xCAFEDECAF/VanBus #4: Are AliExpress's SN65HVD230 boards any good?

Vincent0690, 2022-08-14:
hey
I'm looking for a cheap SN65HVD230 board. My habitual retailer doesn't sell thoses. I found many board for 1-2€ on AE, Amazon list them at 5 to almost 20€.
What link would you recommend?

0xCAFEDECAF, 2022-08-14:
I ordered this one: https://webshop.domoticx.nl/index.php?route=product/product&product_id=3935 . A bit more expensive than AE but not as much as you found at Amazon. Delivery was within a few days, but I'm not sure if they will ship to your country.

It works fine. If you have more time you might just as well order at AE. My experience with AE is that the quality of these basic electronics is pretty good.

Vincent0690, 2022-08-14:
They do ship, but for 25€. I'll go for AE and wait till October 😭

priceparrot, 2022-10-03:
Can I suggest using these product links (shipped from the netherlands):

https://opencircuit.shop/product/sn65hvd230-can-bus-tranceiver-3.3v
https://opencircuit.shop/product/mcp2551-can-bus-interface-module
https://opencircuit.shop/product/voltage-conversion-module-4-channel-5v-3.3v

You could even throw in the latest Wemos to make the list complete:
https://opencircuit.shop/product/wemos-d1-mini-v4.0-wifi-module

0xCAFEDECAF, 2022-10-04:
Thanks for the tip!

I noticed https://opencircuit.nl also works.

0xCAFEDECAF, 2022-10-14:
Solved

closed 2022-10-14 by 0xCAFEDECAF