0xCAFEDECAF/VanBus #5: Can the ESP respond to a message with the ACK bits?

FortunyPT, 2022-11-05:
Hi!
First, I would like to thank you for this amazing lib! It got me deeper into my car's underworld! :)

Currently I'm in a middle of a modernization program on my Peugeot 206 :)
I'm building a head unit out of a raspberry pi using the ESP8266 to make the data bridge while the OEM radio is running in parallel.
So far so good, everything is working fine, I'm able to read from the bus all the data I want and serial print it, but today I removed the multi-function display to replace it with my own coded version to show fuel consumption and, not only the radio didn't work, but my instrument panel lit up like a Christmas tree!
I was kind of expecting the radio not to work because of the missing ACK of the messages directed to it, but the behavior of the dash caught me by surprise.

So my question is, is there a way to write the ACK bits at the end of the data reception so that this doesn't happen?

0xCAFEDECAF, 2022-11-07:
Thanks! :)

I doubt if sending "ACK" bits will solve your issue. I think the MFD is doing a lot more for the system than just sending an "ACK" bit here and there.

If I would write the code, I would start at the place where the timer is set to wait for the ACK; see around [line 1022 of VanBusRx.cpp](https://github.com/0xCAFEDECAF/VanBus/blob/master/VanBusRx.cpp#L1022) . That timer is currently set to 40 us (5 time slots), but you should set it to 8 us (1 time slot). The function to be called by the timer (as set in `timer1_attachInterrupt(<function>);` must write a "1" bit (see 
[line 120 of VanBusTx.cpp ](https://github.com/0xCAFEDECAF/VanBus/blob/master/VanBusTx.cpp#L120)), then set another 8 us timer, which calls a function which writes a "0" bit (see [line 129 of VanBusTx.cpp](https://github.com/0xCAFEDECAF/VanBus/blob/master/VanBusTx.cpp#L129)).

The function which writes the "1" may need an `if` statement that checks the IDEN value of the currently received packets, if you want to ACK only specific packets.

morcibacsi, 2022-11-07:
@0xCAFEDECAF is right, the display is a "master" device. It is responsible for a lot of things. For example: trip computer reset (when you press the right stalk button), querying trip computer data (fuel consumption, distances) and door statuses from the BSI, for the CDC changer and the radio (turning it on/off and sound settings as well). So as you experienced the radio can't be turned on without the display as it works like this: The radio sends a message that the power button was pressed. The display catches this message, and decides what to do (if the radio was on then it sends a command to turn off, or the other way around). The display also decides which source the radio should switch to (thinking of when it is off and you put on a disc, it will automatically turns on the radio and switches to the CD). When you are listening to the radio refreshing the RDS data is also the responsibility of the display. So... it is a difficult job to replace it. Of course all of this can be replicated, but it requires a lot of trial and error. The easiest is to just hide the display in the glovebox or under the dash.
So it is isn't enough to ACK the messages, you would also have to implement "read type frames".  I wrote a few sentences of the different frame types here: https://github.com/morcibacsi/arduino_tss463_van#message-types but I recommend to read the pages 19-21 and 45-46 of the TSS463 datasheet: http://ww1.microchip.com/downloads/en/DeviceDoc/doc4205.pdf
The cristmas tree effect can be achieved by writing to the VAN bus at the wrong time (a device must stop transmitting immediately when a higher priority message is transmitted, or when it detects a different bit on the bus than it wrote). Low signal state has higher preference than "high state".

0xCAFEDECAF, 2022-11-07:
Indeed. I think the most promising option would be to keep the circuit board (PCB) of the MFD (you might ditch the crappy LCD), as @morcibacsi says, tucked under the dash or somewhere else.

FortunyPT, 2022-11-07:
Thanks for the explanation! I also had a feeling that it wouldn't be as easy as writing the ACK bit, since there are more than 4 wires in the harness connecting to the MFD. But when I'm able, I'll have a look at the the interrupt timer function, even though it is a little bit beyond my knowledge.
To reach my goal of modernizing the car, I only imposed one restriction, I can't do something that I can't undo. So if I want to modify an original part I have to get a spare one in case I want to go back. But I don't think that's a problem, a friend of mine has a wrecked 206 that I can get parts from. And who knows, maybe he will let me experiment with the ACK-modified code on it! :)

closed 2022-12-02 by 0xCAFEDECAF