0xCAFEDECAF/VanBus #6: Question: Is every VAN wires gives the same data?

Vincent0690, 2022-12-12:
Hello,

It may sound like a very confusing formulation.
But from your experience, is every VAN connector endpoint gives the same amount of data?
By that I'm asking if there's multiples VAN bus ? (Ex VAN_0 connects BSI to the Radio and communicates only data relevant for the radio, VAN_2 connects ESP to BSI and only gives data related to ESP/ABS)?
As I understood the VAN communication to the radio gives all data, is it true for every VAN wires?
I can't really have a good access to the radio connector, and the VAN radio is not enabled completly (stock radio was optional on purchase, so they added an aftermarket one)

Thank you,

0xCAFEDECAF, 2022-12-12:
Yes, as far as I know there is only one VAN bus: the "comfort bus". There are multiple connector pins labeled "VAN data", see e.g. https://user-images.githubusercontent.com/31759302/187431278-4341194d-2f27-421e-bf0d-3705427ae11d.jpg that shows the pins also on the MFD type C display.

morcibacsi, 2022-12-12:
There are 3 different VAN buses. The odometer, radio, display, reversing radar, cd changer, A/C ECU is connected to the comfort bus which has a speed of 125 kbps. The airbag ECU and the COM2000 is on CAR1 bus and the alarm ecu is on the CAR2 bus and they have the speed of 62.5 kbps. These are 3 different buses with different purposes and with different data. Of course it doesn't matter where you connect to a specific bus, as every endpoint has access to all the data. But make sure you are on the correct bus.
Here is the topology of a Peugeot 206 (it is pretty similar to all cars with VAN bus, but higher end cars can have more nodes on the bus):
[image]

0xCAFEDECAF, 2022-12-12:
Thanks Peter!

Vincent0690, 2022-12-12:
> There are 3 different VAN buses. The odometer, radio, display, reversing radar, cd changer, A/C ECU is connected to the comfort bus which has a speed of 125 kbps. The airbag ECU and the COM2000 is [...]

Very helpful answer. Thank you a lot.

closed 2022-12-12 by Vincent0690