0xCAFEDECAF/VanBus #7: Writing negative temperatures on VAN Bus

mijanto, 2023-07-01:
Hello, 
how do I write negative temperature on VAN Bus? There is _unsinged_ integer for the data rmtTemperatureBytes[]. 
```
uint8_t rmtTemperatureBytes[] = {0x0F, 0x07, 0x00, 0x00, 0x00, 0x00, Temperature * 2 + 0x50};
```
As I set `Temperature` variable to negative number, it is written as positive in that array.

mijanto, 2023-07-01:
Ok, theoretically I figured it out, still needs to be tested. 
Calculation of the number in array was just slightly edited. _Theoretically_ should work up to -40°C, so the variable would be 0 (-40 * 2 + 0x50 = 0).
Edited formula:
```
uint8_t rmtTemperatureBytes[] = {0x0F, 0x07, 0x00, 0x00, 0x00, 0x00, ((Temperature * 2) + 0x50)};
```

0xCAFEDECAF, 2023-07-03:
Hi Mijanto, thanks for noticing! I pushed a fix in commit 0285958 .