0xCAFEDECAF/VanBus #10: MCP2515 SPI

Antonzlo, 2023-12-10:
Hi, is there a native way to use it? Or does it require to rewrite half of the library? 🙂

0xCAFEDECAF, 2023-12-12:
You mean this one: https://ww1.microchip.com/downloads/en/DeviceDoc/MCP2515-Stand-Alone-CAN-Controller-with-SPI-20001801J.pdf ? That is for CAN bus. It will not work for VAN bus because the layout of the bits is different.

Antonzlo, 2023-12-12:
Okay, thanks. But 2551 works? How to know what else works?

0xCAFEDECAF, 2023-12-13:
Yes, [MCP2551 works](https://github.com/0xCAFEDECAF/VanBus#schematics). Also, the SN65HVD230 transceiver works. You could also try the [Atmel TSS463/TSS461 VAN Datalink Controller](http://ww1.microchip.com/downloads/en/DeviceDoc/doc4205.pdf), but it needs a different library which is found here: https://github.com/morcibacsi/arduino_tss463_van . They are difficult to find for sale.

0xCAFEDECAF, 2024-03-07:
Long time no hear; I guess this ticket can be closed.