0xCAFEDECAF/VanBus #12: Program crashes after 2-3 VAN packets with ESP32

unpapardo, 2024-05-13:
Well, pretty much what the title says. The programs starts out OK, and after 1-4 (usually 1 or 2) VAN packets it crashes with abort() and a corrupted backtrace. It can crash mid-print sometimes

14:54:06.417 -> abort() was called at PC 0x40084f1b on core 1
14:54:06.417 -> Backtrace: 0x40083851:0x3ffbeeec |<-CORRUPTED
(You can check the attachment, its all the same)

I don't really know where to start looking for

For the record, here are the board settings:
[image]

Arduino IDE is 2.3.2, esp32 library is 2.0.16 and VanBus library is 0.4.0
Car is a 2007 Peugeot 206, RD3 radio
ESP32 is connected via 10k resistors to VANH, as I dont have a MCP2551 yet. Maybe half the packets do come as CRC corrupted, but I don't think that's the issue here

0xCAFEDECAF, 2024-05-14:
Hi,

What program are you using?

My advice would be to start where I left off. I am using [Arduino IDE 1.8.19](https://downloads.arduino.cc/arduino-1.8.19-windows.exe), and [ESP32 library 1.0.6](https://github.com/espressif/arduino-esp32/releases/tag/1.0.6). Furthermore, board settings seem to be [slightly different](https://github.com/0xCAFEDECAF/VanBus/blob/master/extras/Arduino%20IDE/Board%20settings%20ESP32.png): in my setup, "_Core Debug Level_" is "_None_".

Also, I am not sure if 10 kOhm is ok. My setup uses 4.7 kOhm, but maybe even lower is better. Note that ESP32 GPIO pins have [internall pull-up / pull-down of 45 kOhm](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/gpio.html#internal-pullup-and-pulldown), so a 10 kOhm divider might not make enough voltage difference.

unpapardo, 2024-05-14:
I think some things changed between esp library V1 and V2, so I'll try downgrading to that and let you know.

The resistor divider might have to do with the packet integrity, but I'm pretty sure that's not the case here (or at least not the main one)

unpapardo, 2024-05-29:
Well, after some experimenting and with a MCP2551 (not that it made much difference though), with ESP32 v1.0.6 I get about a 20% CRC_ERROR rate, but it doesn't crash

With v2.0.16, even using the simplest loop possible 
```
void loop() {
    TVanPacketRxDesc pkt;
    if (VanBusRx.Receive(pkt)) pkt.DumpRaw(Serial);
}
```
it still mostly crashes at 1-3 packets. Curiously, sometimes it can get to 50 or 70 pacekts without crashing, though most of them are corrupted

Anyways, I got myself a TSS463, so I'll try my luck with that
Thanks anyways

morcibacsi, 2024-05-30:
> Anyways, I got myself a TSS463, so I'll try my luck with that Thanks anyways

If you just want to read, you don't need the TSS463, a CAN transceiver is enough, and you can try my library for the ESP32 from here: https://github.com/morcibacsi/esp32_rmt_van_rx

If you also want to send packets, then you can use the TSS463, or you can try this library with just a CAN transceiver: https://github.com/morcibacsi/esp32_ulp_van_tx (however this doesn't support all packet types)

Sorry @0xCAFEDECAF I don't want to steal the users of your library, just showing an alternative solution which they can experiment with.

0xCAFEDECAF, 2024-05-30:
No problem @morcibacsi , I am also curious what is the underlying issue here.

@unpapardo I'm afraid I cannot reproduce your problem with the hardware setups I have available. I am using using a [LilyGO TTGO T7 Mini32](https://www.tinytronics.nl/en/development-boards/microcontroller-boards/with-wi-fi/lilygo-ttgo-t7-mini32-v1.3-esp32-4mb-flash) (ESP32 based) and it works flawlessly.

It may help to connect a logic analyser to the ESP board's GPIO pin, which you are using as RX input. In this way you could check what signal levels and signal timing is coming into the ESP32 board.

unpapardo, 2024-05-30:
Thanks guys but I think the TSS463 is my best shot right now (and I dont have a logic analizer, sadly). 
I'm retrofitting Bluetooth into the old RD3 radio* and even if I get this working, BT sinking is fairly process-intensive and I think it will mess with the delicate bit timings on the VAN bus, and I also need ESP32 v2.0+ for those libraries. Plus, eventually writing track info onto the display would be a neat bonus.
Now, is all this worth it just so I can skip songs with the wheel buttons? Debatable, but I'm too deep into this to stop now ahaha

*Btw, if anyone else happens to come across this thread with similar ideas, injecting digital audio to the RD3 radio is pretty simple, without anything VAN related

morcibacsi, 2024-05-31:
@unpapardo if you want to add bluetooth support either you can emulate a CD changer, or I saw somewhere that people are tearing apart the RD3 and tapping the signal directly to the PCB. If yo use the CD changer approach (I have something related to this in the examples on my repos somewhere). In this case you can display only numeric track info like DISC 1(-6) - TRACK 1(-99). If you want to display song names or stuff like that RDS info might be doable, but the display only receives RDS info when the RD3 is in radio mode (AM, FM). But if you would like to fake that it most likely would interfere with the packets generated by the radio.

@0xCAFEDECAF I have both of those analyzers, I used them during my projects. They work fine with the Saleae software. I never tried Pulseview/Sigrok

unpapardo, 2024-05-31:
> `I dont have a logic analizer, sadly`
> 
> What about [this one](https://www.tinytronics.nl/en/tools-and-mounting/measuring/oscilloscopes-and-logic-analyzers/logic-analyzer-8-channel-usb)? It seems to work with [Saleae software](https://www.saleae.com/pages/downloads), and in the reviews it is claimed to also work with [Pulseview/Sigrok](https://sigrok.org/doc/pulseview/0.4.1/manual.pdf).
> 
> But maybe 24 MHz is not fast enough (I have no experience with it). [This one](https://nl.aliexpress.com/item/1005006856304699.html?spm=a2g0o.productlist.main.11.67ec3764vTfCrI&algo_pvid=d292efed-4b54-40b9-b834-5561ef71d328&algo_exp_id=d292efed-4b54-40b9-b834-5561ef71d328-5&pdp_npi=4%40dis%21EUR%2185.39%2161.48%21%21%2190.71%2165.31%21%402103871e17161896818733422e8b22%2112000038527929663%21sea%21NL%21737336263%21&curPageLogUid=pajvPulrnsjI&utparam-url=scene%3Asearch%7Cquery_from%3A) is 100 MHz but a lot more expensive.
> 
Ohhhh that's interesting, and for the price I'm probably gonna get one

> `injecting digital audio to the RD3 radio is pretty simple` 
> 
> Do you have a link or URL? 😄 

> @unpapardo if you want to add bluetooth support either you can emulate a CD changer, or I saw somewhere that people are tearing apart the RD3 and tapping the signal directly to the PCB.
No, I found out by testing and poking stuff that if you turn on the radio with a disk in the CD player (internal one, not CDC), the "CD present" flag doesn't get changed until the eject button is pressed or the radio loses power

We can leverage that fact by then removing the CD player module and injecting S/PDIF audio (with an ESP32 for example) on pin 3 of the module ribbon cable. The radio still enters CD mode without the module and plays back our hijacked audio, and because we are altering the input rather than the output as in most instructions, we still have native volume control👌
(But no song skipping or info, of course)

> If yo use the CD changer approach I have something related to this in the examples on my repos somewhere. In this case you can display only numeric track info like DISC 1(-6) - TRACK 1(-99). If you [...]

Mmmmmh now that's inconvenient 1

closed 2025-06-18 by 0xCAFEDECAF