0xCAFEDECAF/VanBus #13: simulate CDC on autoradio Peugeot RD1 Clarion PU-2294

bagou9, 2024-12-30:
Hello,

I try to simulate changeur CDC on autoradio Peugeot RD1 Clarion PU-2294...
I use Wemos D1 mini + MCP2551 + level shifter.

(I also tested library of @morcibacsi with arduino uno + TSS463, but without good succes).

I succes dump frame on VAN bus with Yatour box (compatible with RD1).
2 frames are important and sent every second by Yatour:
0x4EC (id CDC) and 0x8C4

I trying to re-use this frames in arduino sketch to simulate CDC.
I succes to send this, and autoradio understand them, but only on 2-3 seconds:
On lcd display, Track time increment few seconds, 
and after autoradio display "no changer" and switch in radio mode.

When i compare dump with yatour, and dump with simulate CDC by sktech, I have the impression that an acknowledgment is missing in my sent frames.

When switch src CDC on Autoradio, he start by send frame 0x8EC, flag 0x0C (WA0), data 0x11:0x81, and dump indicate "no ACK".
With Yatour connected, I have not this frames.
I think i must ACK this frames, but I can't find how to do it with your library and I no trace in dump.

Other test:
Yatour connected, autoradio switch src CDC: it's ok, yatour playing and autoradio display track time...
then I disconnect the Yatour, and my sketch take relay and send frames 0x4EC for simulate CDC with data, and 0x8C4, flag 0x0C, data 0x96.
Autoradio continue to display correctly info of CDC, track time continue without problem.
So the ACK of first frames is apply just on start of sequence when switch src.

in attachment 2 files;
- dump with Yatour + CDC simulate,
[dump RD1 yatour and CDC simul.txt](https://github.com/user-attachments/files/18275302/dump.RD1.yatour.and.CDC.simul.txt)

- dump simulate CDC of skecth.
[dump RD1 CDC simul.txt](https://github.com/user-attachments/files/18275402/dump.RD1.CDC.simul.txt)

So my questions:
- Does my reasoning seem correct?
- How to acknowledge particular frame ?
- on "dump RD1 CDC simul", i don't undertand the frame 4EC F (RA1) that receive every seconds when send frame 4EC with data (this frame not exist with yatour).
   "R" = Read; "A" = requesting Ack; "1" = RTR, but why ??

thank you for your help

PS: I am French user :)

morcibacsi, 2024-12-30:
In dump RD1 CDC simul.txt I don't see 8C4 message, but 8EC. Please upload your sketch so I can check 4EC as well.

bagou9, 2024-12-30:
Hello @morcibacsi  :)
I’m happy to see you again on this topic :)

Sorry for file, I updated this  in first message with frame send 8C4....
And I added a excel file compare between esp8266 transmit frames, and esp8266 in sniffer mode.
and my sketch (again in draft mode... )

thank you for your help

morcibacsi, 2024-12-30:
I checked your code. The way you send the 4EC message is wrong. As I see you send it with _VanBus.SyncSendPacket_ and it creates a "Normal frame" on the bus. 
A VAN frame is built by the following: SOF - IDENTIFIER - COM - DATA - CRC - EOD - ACK - EOF

In case of a "Normal frame" all of the above is generated by one device. 

But the 4EC message should be a "Reply Request frame" or "Remote frame". In this case the SOF - IDENTIFIER - COM .... ACK part of the message is generated by a requesting module (or initiator) and the DATA - CRC - EOD - EOF part of the message is generated by te requested module. (side note: an interesting part is the COM field as the requesting module sets it to F but it becomes an E on the bus as the requested module sends a 0 which is dominant on the bus)

In case of the 4EC the initiator is the display, and the requested module is the CDC. So you only need to send the DATA - CRC - EOD - EOF part when you detect 0E 4EC F on the bus.
I browsed the code of the VanBusTx.cpp in the repository, but I couldn't find a method which can generate such a frame, but Erik might correct me if I'm wrong.

I attached a picture where I color coded which part is sent by which participant for better understanding.
[image]

bagou9, 2024-12-30:
thank you for these explanations.
but I don't understand some things...
In dump file yatour, frame 4EC F doesn't exist.
And yatour start and send directly frame 4EC with data.
(I had a similar dump when trying with your esp32_rmt van reader library, cf eevblog forum)

otherwise compared to your picture, if I understand correctly:
I must include the ack of initiator in Reply frame ?

morcibacsi, 2024-12-31:
> In dump file yatour, frame 4EC F doesn't exist. 

You only find 4EC F when there is no response from the CDC. This frame is partially generated by the EMF - it starts the transmission, and tries to write 4EC F on the bus. But as soon as the CDC answers, the 4EC F "turns into" 4EC E (the lowest 1 bit turns into a 0.  F in binary equals to 1111 and E in binary equals to 1110) - as I wrote earlier 0 is dominant on the bus, so the CDC changes the bus state.

> And yatour start and send directly frame 4EC with data. 

By observing the bus you can't really see which module is writing a particular part of the frame, and you can assume that a frame is written by one peripheral, but in some cases (like this one we are discussing) it is actually generated by two modules (first part is EMF second is CDC)

Actually.. in _RD1 CDC simul.txt_ I can see a line 4EC F (RA1) 97-68 this is sent by the display. This should be answered by your sketch and this would "turn into" 4EC E. But as I mentioned, as far as I know you can't do that with this library.

You can check these, like this: every time you send 8C4|C|96 from your sketch 4EC F 9768 will be written by the display.  (just comment  out this line: result = VanBus.SyncSendPacket(0x4EC, 0x0E, packet, 12);)

bagou9, 2024-12-31:
Hello,
Ok. Thanks you again.
Van bus appears is a little more complicated that Can bus... 😉

I will retry others tests with your library van tss463 and Arduino nano.

0xCAFEDECAF, 2024-12-31:
> I browsed the code of the VanBusTx.cpp in the repository, but I couldn't find a method which can generate such a frame, but Erik might correct me if I'm wrong.

Indeed, such method is not yet implemented. I'm afraid that due to personal circumstances I am not able to work on this the coming weeks. But if you like, feel free to implement it yourself in the mean. If you create a pull request I will process it!

bagou9, 2025-01-12:
Hi @morcibacsi ,
I continue my tests with your library TSS463...
I open a discussion on your github https://github.com/morcibacsi/arduino_tss463_van/issues