0xCAFEDECAF/VanBus #14: Error on compilation - Can't use your lib

NzoSifou, 2025-02-26:
Hi!

I can't compile my code because of usage of the old API (2.X) of the Arduino ESP32 core.
Here's the error:

`C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp: In function 'void FinishPacketTransmission(TVanPacketTxDesc*)':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:38:5: error: 'timerAlarmDisable' was not declared in this scope
   38 |     timerAlarmDisable(timer);
      |     ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp: In static member function 'static void TVanPacketTxQueue::StartBitSendTimer()':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:239:11: error: 'timerAlarmEnabled' was not declared in this scope
  239 |     if (! timerAlarmEnabled(timer))
      |           ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp: In function 'void SetTxBitTimer()':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:530:5: error: 'timerAlarmDisable' was not declared in this scope
  530 |     timerAlarmDisable(timer);
      |     ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:541:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  541 |         timerAttachInterrupt(timer, VanBusRx.txTimerIsr, true);
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal.h:98,
                 from C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/Arduino.h:36,
                 from C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.h:29,
                 from C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:12:
C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:542:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  542 |         timerAlarmWrite(timer, VanBusRx.txTimerTicks, true);
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:242:9: error: 'timerAlarmDisable' was not declared in this scope
  242 |         timerAlarmDisable(timer);
      |         ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:243:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  243 |         timerAttachInterrupt(timer, &SendBitIsr, true);
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal.h:98,
                 from C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/Arduino.h:36,
                 from C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.h:29,
                 from C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBus.h:33,
                 from C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:11:
C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:543:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  543 |         timerAlarmEnable(timer);
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:244:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  244 |         timerAlarmWrite(timer, VAN_BIT_TIMER_TICKS, true);
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp: In function 'void RxPinChangeIsr()':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:829:13: error: 'timerAlarmDisable' was not declared in this scope
  829 |             timerAlarmDisable(timer);
      |             ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusTx.cpp:245:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  245 |         timerAlarmEnable(timer);
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1148:13: error: 'timerAlarmDisable' was not declared in this scope
 1148 |             timerAlarmDisable(timer);
      |             ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1149:33: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
 1149 |             timerAttachInterrupt(timer, &WaitAckIsr, true);
      |             ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1150:13: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
 1150 |             timerAlarmWrite(timer, 40 * 5, false); // 5 time slots = 5 * 8 us = 40 us
      |             ^~~~~~~~~~~~~~~
      |             timerWrite
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1151:13: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
 1151 |             timerAlarmEnable(timer);
      |             ^~~~~~~~~~~~~~~~
      |             timerAlarm
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp: In member function 'bool TVanPacketRxQueue::Setup(uint8_t, int)':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1202:23: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
 1202 |     timer = timerBegin(0, 80 / 5, true);
      |             ~~~~~~~~~~^~~~~~~~~~~~~~~~~
C:\Users\nzosi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1203:5: error: 'timerAlarmDisable' was not declared in this scope
 1203 |     timerAlarmDisable(timer);
      |     ^~~~~~~~~~~~~~~~~
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp: In member function 'void TVanPacketRxQueue::Disable()':
C:\Users\nzosi\Documents\Arduino\libraries\VanBus\src\VanBusRx.cpp:1246:5: error: 'timerAlarmDisable' was not declared in this scope
 1246 |     timerAlarmDisable(timer);
      |     ^~~~~~~~~~~~~~~~~
exit status 1
`

There is a migration guide to resolve theses problems, here's the link : https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#timer
I will try to do it myself with a PR but I don't know if I can achieve that :(
At least you are aware of the problem

0xCAFEDECAF, 2025-02-26:
Yes, it compiles for ESP32 up to version 1.0.6 . Not even 2.x or 3.x . I am working on support for the newer versions but time is scarce 😞

NzoSifou, 2025-02-27:
I founded a work around for now, I'm using PlatformIO and it's working great, because PlatformIO havent upgraded to 3.x, it's still on 2.x.

So it compile on 1.x and 2.x 👍 ! Just not in 3.x ^^

I'm not really good in C++ Arduino coding but I'm trying to contribute to your project by switching to 3.x compatibility but it's hard for me 😂

0xCAFEDECAF, 2025-02-28:
> So it compile on 1.x and I thought 2.x too ? Just not in 3.x ^^
Yes, I checked: it compiles on 1.x and 2.x, just not on 3.x . In 3.x they made a lot of non-backward compatible changes...

NzoSifou, 2025-02-28:
Yeah I saw that too...
But, here you go : #15
I tried to make a compatible code to compile it in 3.x :D
Go check the description of the PR please ^^

NzoSifou, 2025-02-28:
Oh and I saw that you have a 406 coupé and I'm trying to use your work in the same car!
But I've tried and it didn't work 😭
Can I get in touch with you in any chat platform please? Like Discord, Snapchat, Instagram, ...?

0xCAFEDECAF, 2025-03-01:
I no longer have the coupé. But I have a 406 estate which also has the VAN bus.
I'm afraid I am not open for chat. I must keep my focus on other things. But I will respond here as soon as time is given to me.

NzoSifou, 2025-03-01:
I totally understand, no worries, thank you already for still taking care of this repository and answering me when you can!