0xCAFEDECAF/VanBus #16: Not able to read VAN network 206+

Antho91, 2025-08-09:
Hi, I wanted to use your project to sniff the audio controls on the VAN network (of a 206+) so that I can translate them for my aftermarket radio.

I'm using a ESP32 board and using port IO21 and IO22 for the CTX and CRX, like u have draw in schematic. I have removed the 2 100Ohm resisters near the CANH and CANL (I also haven't bridged them ;) ).
Also, my R1 (which in your schematic is a 470Ohm resister), I have replaced this with 2 x 220Ohm resisters (because I don't have a 470Ohm res.. Just to mention this).

The issue is that I don't get correct data back (no CRC_OK, only CRC_ERROR)

With only pin 3 connected, I get:
```
17:46:01.314 ->                                           w  u  ·  ·  · 
17:46:01.314 -> Raw: #4368 ( 4/15) 14(19) 0E 5E5 B (R-1) 77-77-77-77-77-77-77-77-5F-FF-F7-77-77-77:BB-E6 NO_ACK OK 7434 CRC_ERROR uBit=24
17:46:01.314 ->                                           w  w  w  w  w  w  w  w  _  ·  ·  w  w  w 
17:46:01.314 -> Raw: #4369 ( 5/15)  6(11) 0E 5DD B (R-1) B7-77-FF-F6-BE-FF:FF-BE NO_ACK ERROR_NBITS 6562 CRC_ERROR uBit=28
17:46:01.346 ->                                           ·  w  ·  ·  ·  · 
17:46:01.346 -> Raw: #4370 ( 6/15)  6(11) 0E B65 B (R-1) 77-77-77-77-77-77:77-5D NO_ACK ERROR_NBITS 5832 CRC_ERROR uBit=12
17:46:01.346 ->                                           w  w  w  w  w  w 
17:46:01.346 -> Raw: #4371 ( 7/15)  2( 7) 0E 5E5 D (WA1) 77-77:E5-7F NO_ACK ERROR_NBITS 4ED2 CRC_ERROR uBit=29
17:46:01.378 ->                                           w  w 
17:46:01.378 -> Raw: #4372 ( 8/15) 10(15) 0E 5DB B (R-1) FF-6E-FB-5F-BB-BB-DF-77-77-77:BF-DD NO_ACK ERROR_NBITS D66E CRC_ERROR uBit=11
17:46:01.378 ->                                           ·  n  ·  _  ·  ·  ·  w  w  w 
17:46:01.378 -> Raw: #4373 ( 9/15)  2( 7) 0E BD5 D (WA1) 77-B7:DD-E6 NO_ACK ERROR_NBITS E9C0 CRC_ERROR uBit=36
17:46:01.378 ->                                           w  · 
17:46:01.411 -> Raw: #4374 (10/15)  5(10) 0E 565 D (WA1) 77-75-FF-FF-F7:E7-BB NO_ACK ERROR_NBITS C370 CRC_ERROR
17:46:01.411 ->                                           w  u  ·  ·  · 
```

With both the pins correctly positions I get (so CANH on pin 3 and CANL on pin 2) I get:
```
17:45:58.083 -> Raw: #4194 (10/15)  0( 3) 0E 4DC 8 (W-0) NO_ACK ERROR_NBITS 0000 CRC_ERROR
17:45:58.083 ->                                          
17:45:58.083 -> Raw: #4195 (11/15)  0( 3) 0E 4D4 F (RA1) NO_ACK ERROR_NBITS 0000 CRC_ERROR
17:45:58.116 ->                                          
17:45:58.116 -> Raw: #4196 (12/15)  0( 3) 0E 664 C (WA0) NO_ACK ERROR_NBITS 0000 CRC_ERROR
17:45:58.116 ->                                          
17:45:58.116 -> Raw: #4197 (13/15)  0( 3) 0E 464 D (WA1) NO_ACK OK 0000 CRC_ERROR uBit=24
17:45:58.116 ->                                          
17:45:58.116 -> Raw: #4198 (14/15)  0( 2) 0E NO_ACK ERROR_NBITS 0000 CRC_ERROR uBit=20
17:45:58.148 ->                                          
17:45:58.148 -> Raw: #4199 (15/15)  0( 2) 0E NO_ACK ERROR_NBITS 0000 CRC_ERROR
17:45:58.148 ->                                          
17:45:58.180 -> Raw: #4200 ( 1/15)  0( 2) 0E NO_ACK ERROR_NBITS 0000 CRC_ERROR uBit=20
17:45:58.180 ->             
```   

Next, I've downloaded this project and copied it over the VANBUS plugin which I've downloaded via Ardiuno IDE, because I got an error (TVanPacketRxQuee::Disabled();....... 'timerAlarmDisable' was not declared in this scope).

Script what I'm using is just as easy as posible:
```
#include <VanBus.h>

const int TX_PIN = 21; // VAN bus transceiver input (ESP32 naar bus)
const int RX_PIN = 22; // VAN bus transceiver output (bus naar ESP32)

void setup() {
  Serial.begin(115200);
  TVanBus::Setup(RX_PIN, TX_PIN);
}

void loop() {
  TVanPacketRxDesc pkt;
  if (VanBus.Receive(pkt)) {
    pkt.DumpRaw(Serial);  // Print ontvangen VAN-pakket
  }

  // Testpakket sturen (optioneel)
  /*
  uint8_t rmtTemperatureBytes[] = {0x0F, 0x07, 0x00, 0x00, 0x00, 0x00, 0x70};
  VanBus.SyncSendPacket(0x8A4, 0x08, rmtTemperatureBytes, sizeof(rmtTemperatureBytes));
  delay(1000);
  */
}
```

Some pictures click  [here](https://nextcloud.itsbas.nl/s/nk2e2JGqWz2NTR2)

0xCAFEDECAF, 2025-08-11:
Hi,

"With both the pins correctly positions " --> that is certainly the way to go. I can recognize VAN bus packet IDENs (4DC, 4D4, 664). But the packets seem to be truncated directly after the first 2 or 3 bytes. Normally there are more bytes in each packet; at least 5 (no data, just CRC), more if there are data bytes. For an example, [have a look here](https://github.com/0xCAFEDECAF/VanBus/blob/master/examples/PacketParser/example_log.txt).

Is there a wiring issue? I.e. do you have a factory-original radio, and if so, is it working correctly?

Further, it seems to me you have mixed more than one version of the ESP32 board package. The fact that you are getting a compilation error like _"'timerAlarmDisable' was not declared"_ can be caused by an inconsistency in the board package file set. I advise you to remove the board package completely, not only from within the Arduino IDE but also with a file manager. On my setup it is located at c:\Users\<my_user_name>\AppData\Local\Arduino15\packages\esp32 ; just remove the esp32 folder completely. Then use the Arduino IDE to install the latest version of the ESP32 board package (3.3.0 seems to be the latest).

Personally I have the best experience with ESP8266-based boards. Somehow, the firmware has less interrupt latency than that of the ESP32 based boards.

I see you live in The Netherlands; if you like you are welcome to come to my place so we can experiment a bit and see if we get it working. We can compare your setup with mine. Give me your e-mail address if you like.

Success with the tinkering!

Antho91, 2025-08-12:
Nice to know that you are also from the Netherlands.

I will first (if the weather is less warm) do some tinkering on my own. If that does not work, then I will contact you ;)

I did not yet check without the RX functionality, so I will check that out, The mixed version board packages. That could also be an issue, but with the default Arduino VANBUS library, I had those issues. I will remove everything and rebuild the 2 libraries.

After I've copied your repository into the VANBUS library folder, only then I could compile the code (without the errors like _'timerAlarmDisable' was not declared"_). But I first will remove everything and validate that everything is working,

I also have some ESP8266 boards lying around, So I could also easily switch to that platform if needed. But for now I prefer the ESP32 board.

Antho91, 2025-09-13:
Oke I've just tried it out with a nodemcu v3 ESP8266 board and I get the same error's

```
15:04:56.048 -> Raw: #6394 ( 5/15)  0( 4) 0E 8A4 8 (W-0) AE NO_ACK ERROR_NBITS DB9E CRC_ERROR uBit=4
15:04:56.048 ->                                          
15:04:56.048 -> Raw: #6395 ( 6/15)  0( 4) 0E 824 8 (W-0) 10 NO_ACK OK 224E CRC_ERROR uBit=8
15:04:56.048 ->                                          
15:04:56.048 -> Raw: #6396 ( 7/15)  0( 4) 0E 4FC C (WA0) 80 NO_ACK ERROR_NBITS 121C CRC_ERROR uBit=4
15:04:56.081 ->                                          
15:04:56.081 -> Raw: #6397 ( 8/15)  0( 5) 0E 824 8 (W-0) 11-10 NO_ACK OK F9C0 CRC_ERROR uBit=2
15:04:56.114 ->                                          
15:04:56.114 -> Raw: #6398 ( 9/15)  0( 3) 0E 464 C (WA0) NO_ACK ERROR_NBITS 0000 CRC_ERROR uBit=4
15:04:56.114 ->                                          
15:04:56.147 -> Raw: #6399 (10/15)  0( 4) 0E 824 8 (W-0) 10 NO_ACK OK 224E CRC_ERROR uBit=3
15:04:56.147 ->                                          
15:04:56.147 -> received pkts: 6400, corrupt: 6398/6400 (99.969%), repaired: 70 (1%) [DB:2, SB:51, DCB:17, DSB:0], UCB:50, overall: 6328 (98.875%), addBitTime: 8, maxQueued: 5/15
15:04:56.179 -> Raw: #6400 (11/15)  0( 4) 0E 4DC C (WA0) 80 NO_ACK ERROR_NBITS 2C68 CRC_ERROR uBit=4
15:04:56.179 ->                                          
15:04:56.179 -> Raw: #6401 (12/15)  0( 4) 0E 5E4 C (WA0) 00 NO_ACK OK E1BC CRC_ERROR uBit=2
15:04:56.179 ->                                          
15:04:56.179 -> Raw: #6402 (13/15)  1( 6) 0E 664 C (WA0) 00:10-10 NO_ACK ERROR_NBITS 75F0 CRC_ERROR uBit=36
15:04:56.211 ->                                           · 
15:04:56.211 -> Raw: #6403 (14/15)  0( 5) 0E 824 8 (W-0) 11-10 NO_ACK OK F9C0 CRC_ERROR uBit=2
15:04:56.211 ->                                          
15:04:56.211 -> Raw: #6404 (15/15)  0( 5) 0E E24 8 (W-0) 56-5A NO_ACK ERROR_NBITS CDAE CRC_ERROR uBit=2
15:04:56.211 ->                                          
15:04:56.211 -> Raw: #6405 ( 1/15)  0( 3) 0E 464 C (WA0) NO_ACK ERROR_NBITS 0000 CRC_ERROR uBit=4
15:04:56.244 ->                                          
15:04:56.244 -> Raw: #6406 ( 2/15)  0( 1) 0E NO_ACK ERROR_NBITS 0000 CRC_ERROR uBit=4
15:04:56.244 ->                                          
15:04:56.244 -> Raw: #6407 ( 3/15)  0( 4) 0E 824 8 (W-0) 10 NO_ACK ERROR_NBITS 224E CRC_ERROR uBit=28
```

Just to clarify, these are my Arduino IDE settings

[image]

0xCAFEDECAF, 2025-09-15:
Sure. My offer is still open to meet up and do some tinkering together. I also have a [digital analyser](https://www.tinytronics.nl/en/tools-and-mounting/measuring/oscilloscopes-and-logic-analyzers/logic-analyzer-8-channel-usb) so that we can trace the signal from the source.

0xCAFEDECAF, 2025-10-01:
Sure. If you post a (temporary) private e-mail address here we can discuss further details.

Antho91, 2026-02-04:
Hi @kamil00110, tbh, I totally forget about this project, I also had some issues with the car (unrelated to this xD), and then life hahah. Uuhm what I recall I put them in serie

0xCAFEDECAF, 2026-02-25:
> In the end I was looking to update the time and date on the info screen, so that I don't need to manually enter it after a battery disconnect or change from winter to summer 😅

That would be cool, however, to the best of my knowledge, there is no VAN bus packet that would be able to manipulate the internal clock/calendar of the multi-functional display.

> What do you mean with in frame response? Is this linked with my clock update function?

See the link I already mentioned: "_[perfectly explained by Peter here](https://github.com/0xCAFEDECAF/VanBus/issues/13#issuecomment-2565933066)_". More explanation [is found here](http://www.educauto.org/files/file_fields/2013/11/18/mux3.pdf#page=5) (in French).

TLDR; VAN bus offers the possibility for a device to request for data, where the queried device sends its response _directly, within the same frame_. As said, this driver does not support that (yet).